ros-industrial/industrial_core #112: Industrial_robot_client: Greedy Watchdog leads to rejection of valid goal

simonschmeisser, 2015-06-03:
Hi,

this is a regression due to #86 / #85 . Sometimes execution of valid trajectories is rejected with this message:
Joint trajectory action rejected: waiting for (initial) feedback from controller

I poll my robot with 20 Hz (arbitrary), the watchdog is set to 1Hz. So I think the following happens:

0s: JointTrajectoryAction::controllerStateCB: trajectory_state_recvd_ = true;
0.01s: JointTrajectoryAction::watchdog: trajectory_state_recvd_ = false;
0.02s: JointTrajectoryAction::goalCB -> rejected
(0.03s: JointTrajectoryAction::goalCB -> rejected)
(0.04s: JointTrajectoryAction::goalCB -> rejected)
0.05s: JointTrajectoryAction::controllerStateCB: trajectory_state_recvd_ = true; but too late!

(the events in brackets are due to me retrying the execution if it doesn't work)

So I think the watchdog should manage some other variable like "controller_alive_" which is then checked by goalCB. trajectory_state_recvd_ is meant as an internal variable to the watchdog.

As a workaround I can of course increase/add a sleep between successive attempts to send the goal.

VictorLamoine, 2015-06-03:
Related: http://answers.ros.org/question/206418/how-should-i-handle-joint-trajectory-action-rejected/

shaun-edwards, 2015-06-03:
Does this issue only occur on startup?

VictorLamoine, 2015-06-03:
Not for me, it happens randomly when sending trajectories.
It happens both in simulation and on the real (Fanuc in my case) robot.

simonschmeisser, 2015-06-03:
No, this doesn't happen on startup (maybe if you manage to send goal requests immediately on startup it would, depending on what trajectory_state_recvd_ gets initialized to ?)

the (made up) timestamps in the explanation are meant to be relative

gavanderhoorn, 2015-06-03:
The described behaviour manifests itself when the goal is received in between the watchdog resetting `trajectory_state_received_` and the `controllerStateCB(..)` setting it to `true`. Depending on the rate at which `FollowJointTrajectoryFeedback` are published and the period of the watchdog, there can be quite some time between the two.

I think a proper fix would be for `controllerStateCB(..)` to reset the watchdog clock, instead of introducing another (time delayed) copy of `trajectory_state_received_`.

Alternatively, the behaviour of the watchdog could be change: it now unconditionally resets `trajectory_state_received_` on every execution.

simonschmeisser, 2015-06-03:
I think of it more like a summary than a copy and then it makes sense to have it, since trajectory_state_received_ is "internal" to the watchdog.

Saving the time of the last package would work as well I guess, but it sounds like more overhead.

gavanderhoorn, 2015-06-03:
> Saving the time of the last package would work as well I guess, but it sounds like more overhead.

I'm not suggesting to save anything, just to have the timer reset on each call to `controllerStateCB(..)`. That is how watchdogs typically work.

shaun-edwards, 2015-06-09:
Am I correct in assuming this discussion is about PR #113?

simonschmeisser, 2015-06-09:
Yes indeed, what I proposed there was the least intrusive fix, but I can also implement what @gavanderhoorn suggested. Will just take a day or two

gavanderhoorn, 2015-06-09:
IIRC, @JeremyZoss wrote these parts of `industrial_robot_client`, perhaps he could explain why he implemented the watchdog like it is?

JeremyZoss, 2015-06-09:
Actually, I "inherited" the `joint_trajectory_action` server from Shaun's [original](https://code.google.com/p/swri-ros-pkg/source/browse/trunk/industrial/motoman/src/joint_trajectory_action.cpp?spec=svn40&r=40) code.  So, I can't provide any clarification on the original design intent.

But I will add my +1 to @gavanderhoorn's suggestion to reset[1] the timer in `controllerStateCB`.  If the watchdog callback ever triggers, it means that no trajectory was seen and the goal should be canceled.  I think we could probably get rid of `trajectory_state_recvd_`.

[1] [`ros::Timer`](http://docs.ros.org/indigo/api/roscpp/html/classros_1_1Timer.html) doesn't support a `reset` method directly.  You must call `timer.stop()` followed by `timer.start()` to reset.  See [here](http://answers.ros.org/question/11308/rostimer-will-stopping-it-with-stop-cause-its-counter-to-restart/).

simonschmeisser, 2015-06-09:
Please see the updated PR #113 , I didn't see a way to check if the timer is running, so I kept the controller_alive_ variable I introduced. I can squash these two commits once we decided which one to take

JeremyZoss, 2015-06-09:
+1.  Looks good!

This version will generate messages at 1Hz from Watchdog callback (if not connected to robot), which is the same as the current behavior.  We could set the `oneshot` option in [`createTimer`](http://docs.ros.org/indigo/api/roscpp/html/classros_1_1NodeHandle.html#ac45b36c542acfc90723f7886beb97376) to limit message spam, if desired.

simonschmeisser, 2015-06-10:
I changed it to oneShot and increased the severeness of the notification to WARN so that we actually notice that we are no longer spammed/informed

simonschmeisser, 2015-06-16:
Now I squashed the fixes into one commit, so it looks better on the commit log

gavanderhoorn, 2015-07-10:
As #113 is merged, can this be closed @simonschmeisser?

PR #113 (2015-06-25): Fix rejected actions when goal requests arrive inbetween watchdog reset und arrival of next status package
this fixes #112  , a regression due to #86 / #85 . Sometimes execution of valid trajectories is rejected with this message:
Joint trajectory action rejected: waiting for (initial) feedback from controller

I poll my robot with 20 Hz (arbitrary), the watchdog is set to 1Hz. So I think the following happens:

0s: JointTrajectoryAction::controllerStateCB: trajectory_state_recvd_ = true;
0.01s: JointTrajectoryAction::watchdog: trajectory_state_recvd_ = false;
0.02s: JointTrajectoryAction::goa [... cut]
closed 2015-07-10 by simonschmeisser