ros-industrial/industrial_core #114: Premature termination of trajectory action

Jmeyer1292, 2015-06-15:
Hello all,

The ros-industrial client's action server interface performs checks for 'goal completion' upon accepting a new trajectory and as it gets feedback about the robot state. See [this link](https://github.com/ros-industrial/industrial_core/blob/indigo-devel/industrial_robot_client/src/joint_trajectory_action.cpp#L244). 

Unfortunately this means that any trajectory starting at the 'end point' is immediately considered successfully executed. Furthermore any trajectory that passes through the end point, even if there are more points to go, will be considered successfully executed.

If I'm using the system wrong, please let me know. I'm not sure what the best route to fixing the issue (if it is one) is yet, but I'll look into it in a bit.

VictorLamoine, 2015-06-16:
Happened to me already, I thought I was doing something wrong at that time!

gavanderhoorn, 2015-06-16:
This seems to be at least counter-intuitive, so I think this is not you (@Jmeyer1292) "using the system wrong". You could say that trajectories should be made up of segments that do not overlap, but that seems like an unreasonable restriction.

Perhaps the nr of points should be used to check whether a trajectory *could be* completed at the point where `withinGoalConstraints(..)` returns `true`, but I'm not sure the `joint_trajectory_action` node has access to that information.

shaun-edwards, 2015-06-28:
Address in Indigo by #115

shaun-edwards, 2015-07-28:
This bug was not fixed by #115 or at least there was another manifestation with a different root cause.  In short, motions with the same begin and end point could return complete immediately if the if the `in_motion` status (and corresponding robot motion) did not begin immediately.  In other words, there exists a race condition between the setting and checking of the `in_motion` status.  More discussion on potential solutions can be found in #119.  As #119 is not a proper solution, this issue will remain open.