ros-industrial/industrial_core #122: rosparam initial_joint_state not updating tf

VictorLamoine, 2015-10-30:
I'm providing an initial robot pose when simulating the Fanuc R1000iA:
```yaml
  <rosparam param="initial_joint_state">[-0.1855634061, -0.2878746068, -0.9585348252, 0.1261873049, -1.1276223231, -1.4505605946]</rosparam>
```

However the initial pose of the robot is not updated and is the one when the robot has 0 values on all joints:
```cpp
  ros::init (argc, argv, "test_tcps");
  ros::NodeHandle node;
  tf::TransformListener listener;
  std::string frame = "/base";

  move_group_interface::MoveGroup group ("manipulator");
  Eigen::Affine3d robot_pose;
  listener.waitForTransform ("/base", "/tool0", ros::Time::now (), ros::Duration (1.0));
  tf::poseMsgToEigen (group.getCurrentPose ().pose, robot_pose);
  std::cout << "Current robot pose:\n" << robot_pose.matrix () << std::endl;
```

```bash
Current robot pose:
-2.05103e-10 -8.97932e-11            1         1.51
 6.46235e-27           -1 -8.97932e-11            0
           1 -1.84169e-20  2.05104e-10        1.545
           0            0            0            1
```

gavanderhoorn, 2015-10-30:
`initial_joint_state` is actually a private parameter of the `industrial_robot_simulator` (from [here](http://wiki.ros.org/industrial_robot_simulator)).

Are you using that node? I'm not sure others would actually do anything with it.

VictorLamoine, 2015-10-30:
Yes I'm using the `industrial_robot_simulator`, should I be using something else?!
>  I'm not sure others would actually do anything with it.

Others what?

Minimal example package:
https://github.com/VictorLamoine/initial_joint_state_test

gavanderhoorn, 2015-10-30:
> Yes I'm using the `industrial_robot_simulator`, should I be using something else?!

no, but you do need to set it as a *private parameter* on the node itself, ie:
```xml
<node pkg="industrial_robot_simulator" type="industrial_robot_simulator" name="industrial_robot_simulator">
  <rosparam param="initial_joint_state">[-0.1855634061, -0.2878746068, -0.9585348252, 0.1261873049, -1.1276223231, -1.4505605946]</rosparam>
</node>
```

Or via some other means.

> Others what?

other nodes.

VictorLamoine, 2015-10-30:
I tried:
```xml
  <!-- run the robot simulator and action interface nodes -->
  <group if="$(arg sim)">
<!-- industrial_robot_simulator: accepts robot commands and reports status -->
    <node pkg="industrial_robot_simulator" type="industrial_robot_simulator" name="industrial_robot_simulator">
      <rosparam param="initial_joint_state">[-0.1855634061, -0.2878746068, -0.9585348252, 0.1261873049, -1.1276223231, -1.4505605946]</rosparam>
    </node>
    <node pkg="industrial_robot_client" type="joint_trajectory_action" name="joint_trajectory_action"/>
  </group>
```

But it doesn't even move the robot in RViz!

gavanderhoorn, 2015-10-30:
Ok, so I was mistaken. It's not actually a private parameter (I thought I implemented it as such).

Adding it as a normal parameter through a `rosparam` tag works for me though, when just using the MoveIt RViz plugin (both state display and planning use the initial joint state). Your example pkg does show all zeros.

I suspect your missing some initialisation bits for the move group which cause it to not be updated correctly. I remember similar issues in the past with the MoveIt RViz plugin (see ros-planning/moveit_ros#405 fi).

shaun-edwards, 2015-11-10:
@VictorLamoine, has this issue been resolved?  Can this be closed?

VictorLamoine, 2015-11-10:
I did not solve the issue because I don't understand half of the discussions Gijs pointed out :dizzy_face:

jrgnicho, 2015-11-10:
@VictorLamoine The snippet below works for me:
```
  <group if="$(arg sim)">
    <include file="$(find industrial_robot_simulator)/launch/robot_interface_simulator.launch" />
    <rosparam param="initial_joint_state">[0 ,0.2 ,1.68 ,0.3 ,-0.93 ,0 ,0.785398163]</rosparam>
  </group>
```

VictorLamoine, 2015-11-10:
It doesn't work for me;

```xml
  <!-- run the robot simulator and action interface nodes -->
  <group if="$(arg sim)">
    <include file="$(find industrial_robot_simulator)/launch/robot_interface_simulator.launch" />
    <rosparam param="initial_joint_state">[0.3, 0.9, 0.3, 0.5, -1.57, 0.0]</rosparam>
  </group>
```

I still get the default robot pose and 0 joint values from the `MoveGroup`.

jrgnicho, 2015-11-10:
The robot simulator also looks for the **controller_joint_names** ros parameter that contains a list of the join names:
```
controller_joint_names: ['joint_1', 'joint_2', 'joint_3', 'joint_4', 'joint_5', 'joint_6', 'joint_7']
```

Of course, you'd list the right joints for your robot.

VictorLamoine, 2015-11-10:
This is already provided by [joint_names_m10ia.yaml](https://github.com/ros-industrial/fanuc/blob/indigo-devel/fanuc_m10ia_support/config/joint_names_m10ia.yaml) which is called on the [top of my launch file](https://github.com/VictorLamoine/initial_joint_state_test/blob/master/launch/test.launch#L11).

jrgnicho, 2015-11-10:
Are you using the debian release or the source code?  @shaun-edwards do you know if the change for accepting an initial joint state ever made it to the debian?

VictorLamoine, 2015-11-10:
I'm using `industrial_core` source code (`indigo-devel`).
So you can't reproduce the problem on your machine using my example package?

jrgnicho, 2015-11-10:
Yes, but it doesn't appear to be related to the industrial_robot_simulator node as rviz shows the robot in the correct initial pose and if you echo /joint_states you see the angle values as set in the initial_joint_states parameter:
```
ros-indigo: rostopic echo -n 1 /joint_states 
header: 
  seq: 6311
  stamp: 
    secs: 1447186914
    nsecs: 163320064
  frame_id: ''
name: ['joint_1', 'joint_2', 'joint_3', 'joint_4', 'joint_5', 'joint_6']
position: [0.3, 0.9, 0.3, 0.5, -1.57, 0.0]
velocity: []
effort: []
```

jrgnicho, 2015-11-10:
Not sure why the MoveGroup interface does't provide the current robot status.  An alternative way to get the current joint values is by requesting the planning scene as follows:
```
#include <ros/ros.h>
#include <moveit/move_group_interface/move_group.h>
#include <tf/transform_listener.h>
#include <tf_conversions/tf_eigen.h>
#include <eigen_conversions/eigen_msg.h>
#include <moveit_msgs/GetPlanningScene.h>
#include <moveit_msgs/GetMotionPlan.h>

const std::string GET_PLANNING_SCENE_SERVICE = "get_planning_scene";

int main (int argc,
      char **argv)
{
  ros::init (argc, argv, "ensenso_test");
  ros::NodeHandle node;

  moveit_msgs::GetPlanningScene::Request req;
  moveit_msgs::GetPlanningScene::Response res;
  ros::ServiceClient get_planning_scene_client = node.serviceClient<moveit_msgs::GetPlanningScene>(GET_PLANNING_SCENE_SERVICE);
  req.components.components = req.components.SCENE_SETTINGS | req.components.ROBOT_STATE | req.components.ALLOWED_COLLISION_MATRIX |
      req.components.WORLD_OBJECT_GEOMETRY | req.components.WORLD_OBJECT_NAMES | req.components.OCTOMAP ;

  if(get_planning_scene_client.waitForExistence(ros::Duration(2.0)) && 
    get_planning_scene_client.call(req,res))
  {
    ROS_INFO_STREAM(res.scene.robot_state);
  }
  else
  {
    ROS_ERROR("Service call for planning scene failed");
  }

  return (0);
}
```

gavanderhoorn, 2015-11-10:
This is why I wrote earlier that I think the initialisation of the MoveGroup is missing some things. As @jrgnicho pointed out, 'everything' is up-to-date wrt the joint states, except the move group.

re: initial joint state support in debs: that was released in v0.4.0 in Indigo.

gavanderhoorn, 2017-10-07:
@VictorLamoine: did you ever figure this one out?

I'm going to assume this is a MoveIt issue and close it here.

If you feel that is not correct, please re-open and we can see whether we can fix it here.

closed 2017-10-07 by gavanderhoorn

VictorLamoine, 2017-10-08:
I have never solved this issue, I'll test this again with ROS Kinetic on monday.

VictorLamoine, 2017-10-09:
I have tested with ROS Kinetic and it works, test package is here: https://gitlab.com/InstitutMaupertuis/fanuc_motion

```bash
Current robot pose:
 0.788737  0.517957 -0.331081    1.1717
 0.244385 -0.758388 -0.604255  0.312267
-0.564066  0.395687  -0.72475  0.554187
        0         0         0         1
J0: 0.3
J1: 0.9
J2: 0.3
J3: 0.5
J4: -1.57
J5: 0
```

```bash
$ rosrun tf tf_echo /base /tool0
At time 1507539810.184
- Translation: [1.172, 0.312, 0.104]
- Rotation: in Quaternion [0.904, 0.211, -0.247, 0.276]
            in RPY (radian) [2.642, 0.599, 0.300]
            in RPY (degree) [151.367, 34.337, 17.215]
```

gavanderhoorn, 2017-10-09:
Great. Thanks for testing again.