ros-industrial/industrial_core #129: Indigo-devel branch fails travis ci

shaun-edwards, 2015-12-16:
PR #128 fails build test.  The failure appears to be related to a `roslaunch` test, but the reason is not apparent.  The OSRF build farm does not appear to have this problem, see [here](http://jenkins.ros.org/job/ros-indigo-industrial-core_binarydeb_saucy_amd64/).

130s, 2016-02-08:
@shaun-edwards [said](https://github.com/ros-industrial/industrial_core/pull/135#issuecomment-181443209):

> FYI...it appears the failure may be related to the work-around referenced by ros-industrial/ros_canopen#159

I thought so too and am looking at it now.

gavanderhoorn, 2017-10-07:
Closing this as I don't believe this is a problem any more.

If/when this turns out still to be an issue we can re-open.

closed 2017-10-07 by gavanderhoorn