ros-industrial/industrial_core #132: Crash with industrial_core source

VictorLamoine, 2016-01-21:
I've been using the `industrial_core` source without problem until last week; now the `industrial_robot_simulator` node systematically crashes like this;

```
[industrial_robot_simulator-2] process has died [pid 28201, exit code 127, cmd /home/dell/catkin_workspace/src/industrial_core/industrial_robot_simulator/industrial_robot_simulator __name:=industrial_robot_simulator __log:=/home/dell/.ros/log/50e913b0-c041-11e5-adcd-f01faf263c5d/industrial_robot_simulator-2.log].
log file: /home/dell/.ros/log/50e913b0-c041-11e5-adcd-f01faf263c5d/industrial_robot_simulator-2*.log
```

This is using the `indigo-devel` branch and the latest commit. If I remove the `industrial_core` source from my workspace (thus using the binaries provided from the repository) everything works well.

I did not manage to get logs nor backtrace of the crash at the moment. Here is what I have tried;
- Adding `output="screen"` in the [robot_interface_simulator.launch](https://github.com/ros-industrial/industrial_core/blob/indigo-devel/industrial_robot_simulator/launch/robot_interface_simulator.launch) node. It looks like:

```yaml
  <!-- industrial_robot_simulator: accepts robot commands and reports status -->
  <node pkg="industrial_robot_simulator" type="industrial_robot_simulator" name="industrial_robot_simulator" output="screen"/>
  
  <!-- joint_trajectory_action: provides actionlib interface for high-level robot control -->
  <node pkg="industrial_robot_client" type="joint_trajectory_action" name="joint_trajectory_action" output="screen"/>
```
- Set `CMAKE_BUILD_TYPE`  to `Debug` for the whole catkin workspace. Add `launch-prefix="gdb -ex run --args"` in the launch files. it looks like:

```yaml
  <!-- industrial_robot_simulator: accepts robot commands and reports status -->
  <node pkg="industrial_robot_simulator" type="industrial_robot_simulator" name="industrial_robot_simulator" output="screen" launch-prefix="gdb -ex run --args"/>
  
  <!-- joint_trajectory_action: provides actionlib interface for high-level robot control -->
  <node pkg="industrial_robot_client" type="joint_trajectory_action" name="joint_trajectory_action" output="screen" launch-prefix="gdb -ex run --args"/>
```

But it doesn't work as expected:
```
Type "apropos word" to search for commands related to "word"...
"/home/dell/catkin_workspace/src/industrial_core/industrial_robot_simulator/industrial_robot_simulator": not in executable format: File format not recognized
Starting program:  __name:=industrial_robot_simulator __log:=/home/dell/.ros/log/4e112638-c045-11e5-9d5f-f01faf263c5d/industrial_robot_simulator-2.log
No executable file specified.
Use the "file" or "exec-file" command.
```

I have reproduced this bug on 3 other machine, another machine is not affected.

gavanderhoorn, 2016-01-21:
re: `launch-prefix` with gdb doesn't work: you cannot use that for Python scripts directly like that. In any case, you'd be debugging a `python` instance, not the script itself.

Also: what is the output when running it with `output=screen`?

VictorLamoine, 2016-01-21:
I'll try with `pdb`, `output=screen` still produces empty logs.

gavanderhoorn, 2016-01-21:
Also:
 1. does it crash immediately, or only after some use?
 1. make sure `robot_interface_simulator.launch` is actually being used: I've seen launch files directly start the nodes instead of including that file
 1. `output=screen` does not go to the logs, that's the point. There is no output in your terminal at all?

gavanderhoorn, 2016-01-21:
You can also always use `roslaunch --screen ..` to force all nodes to output to the screen, without having to edit any files.

gavanderhoorn, 2016-01-21:
Exit code `-127` generally means something like 'command not found'.

Can you check you have `/usr/bin/env` on the machines where it crashes? Also, which version of Python are you running? And what is the output of `/usr/bin/env python` on the crashing machines?

VictorLamoine, 2016-01-21:
1. It crashes immediately.
2. I'll check that.
3. The terminal output, not the log files sorry!

```bash
$ roslaunch --screen fanuc_motion cube.launch 
... logging to /home/dell/.ros/log/3f2d8394-c047-11e5-b973-f01faf263c5d/roslaunch-M6700-29527.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://M6700:52991/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_1', 'join...
 * /move_group/allow_trajectory_execution: True
 * /move_group/allowed_execution_duration_scaling: 1.2
 * /move_group/allowed_goal_duration_margin: 0.5
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'j...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/longest_valid_segment_fraction: 0.05
 * /move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /move_group/manipulator/projection_evaluator: joints(joint_1,jo...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/max_nearest_neighbors: 10
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/delay_collision_checking: 1
 * /move_group/planner_configs/RRTstarkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTstarkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/range: 0.0
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/frountierNodeRatio: 0.1
 * /move_group/planner_configs/TRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/TRRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/TRRTkConfigDefault/init_temperature: 10e-6
 * /move_group/planner_configs/TRRTkConfigDefault/k_constant: 0.0
 * /move_group/planner_configs/TRRTkConfigDefault/max_states_failed: 10
[... 252 lines cut ...]
DBUS_SESSION_BUS_ADDRESS=unix:abstract=/tmp/dbus-90Z9stwzST
PKG_CONFIG_PATH=/home/dell/MEGAsync/catkin_workspace/devel/lib/pkgconfig:/home/dell/MEGAsync/catkin_workspace/devel/lib/x86_64-linux-gnu/pkgconfig:/opt/ros/indigo/lib/x86_64-linux-gnu/pkgconfig:/opt/ros/indigo/lib/pkgconfig
LESSOPEN=| /usr/bin/lesspipe %s
INSTANCE=
TEXTDOMAIN=im-config
CMAKE_PREFIX_PATH=/home/dell/MEGAsync/catkin_workspace/devel:/opt/ros/indigo
XDG_RUNTIME_DIR=/run/user/1000
DISPLAY=:0
XDG_CURRENT_DESKTOP=Unity
GTK_IM_MODULE=ibus
LESSCLOSE=/usr/bin/lesspipe %s %s
TEXTDOMAINDIR=/usr/share/locale/
ROS_ETC_DIR=/opt/ros/indigo/etc/ros
COLORTERM=gnome-terminal
XAUTHORITY=/home/dell/.Xauthority
_=/usr/bin/env
```

```
$ /usr/bin/env python
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

gavanderhoorn, 2016-01-21:
```
: Aucun fichier ou dossier de ce type
```
Seems to point to some issue.

Seeing all the `MEGAsync` references: are you syncing yur workspace to a Windows computer as well by any chance? Could you check the line-endings of the `industrial_robot_simulator` script?

VictorLamoine, 2016-01-21:
Got it, looks like a nasty character in one of the files!!

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/pdb.py", line 1314, in main
    pdb._runscript(mainpyfile)
  File "/usr/lib/python2.7/pdb.py", line 1233, in _runscript
    self.run(statement)
  File "/usr/lib/python2.7/bdb.py", line 400, in run
    exec cmd in globals, locals
  File "<string>", line 1, in <module>
  File "/home/dell/MEGAsync/catkin_workspace/devel/lib/industrial_robot_client/joint_trajectory_action", line 1
SyntaxError: Non-ASCII character '\x80' in file /home/dell/MEGAsync/catkin_workspace/devel/lib/industrial_robot_client/joint_trajectory_action on line 2, but no encoding declared; see http://www.python.org/peps/pep-0263.html for details
Uncaught exception. Entering post mortem debugging
Running 'cont' or 'step' will restart the program
> <string>(1)<module>()
(Pdb) 

```

gavanderhoorn, 2016-01-24:
Well, I'm inclined to believe it's something local to your setup(s), as that particular file hasn't seen any changes since 2015-01-16 ([history](https://github.com/ros-industrial/industrial_core/commits/indigo-devel/industrial_robot_simulator/industrial_robot_simulator)).

`0x80` is indeed not in ASCII range, so message is correct. Just where that character is coming from then is hard to understand.

VictorLamoine, 2016-02-02:
My git policy regarding CRLF was set to true (`git config --global core.autocrlf true`), I think this is what caused the problem.

Switching to `git config --global core.autocrlf input`, removing the workspace and configure/compiling agains solves the problem.

I'm not 100% sure because I'm not able to reproduce the bug again by switching to `true` again... :smile:

closed 2016-02-02 by VictorLamoine

gavanderhoorn, 2016-02-02:
Ok, thanks for reporting back.

Good to hear you solved it.

gavanderhoorn, 2016-02-02:
And just to clarify: this is a client-side configuration option. The sources in the repository are ok.