ros-industrial/industrial_core #133: simple_message: add fields to support recovering from a protocol desync

gavanderhoorn, 2016-01-24:
Simple Message as it is right now does not provide any means of recovering from a desync. The only really feasible thing is to do a reconnect cycle.

Most implementations will be running over TCP/IP, so they benefit from the built-in checksumming and in-order delivery guarantees of that protocol, but those do not protect against misbehaving SM implementations or other application level hickups. I recently ran into an example where an implementation did not properly flush its buffers before/after (re)connect of a client, causing the client to receive a partial message. The first byte was immediately interpreted as the `pkt_length`, while in reality it was part of the body. Needless to say that didn't work very well.

If we want to maintain bw-compatibility, we could consider adding two required fields to the bodies of all future messages: one at the beginning, and one at the end. By choosing suitable values, these two could then be used by implementations to fi ensure msg bounderies -- calculated from `pkt_length` -- make sense. The first field would be situated *directly after* the last header field. The second field would come immediately *before* the next `pkt_length`.

Field type: `shared_int`. Possible values: `SMSM` and `MSMS` (ascii chars). BW-compatibility is maintained because only new msgs would use this (and unknown msg IDs should be ignored/NACKed by current implementations) and we could make this user-configurable behaviour (default to `disabled`).

gavanderhoorn, 2016-01-24:
With some thought we could also use these fields to implement an automatic endianess detection scheme perhaps (#80).

shaun-edwards, 2016-01-25:
This seems like a problem that would be common to all comms protocols.  Is there an "accepted" way of handling this?

gavanderhoorn, 2016-01-25:
> This seems like a problem that would be common to all comms protocols. Is there an "accepted" way of handling this?

yes: using special fields with standardised contents so deserialisation state machines can look for them :).

gavanderhoorn, 2016-01-25:
To clarify / summarise:

 - current messages would not be changed
 - new messages are required to include two new fields:
  - one at the start of the body (right after the `reply_type`)
  - the other at the end of the body
 - contents is pre-defined, and should either be:
   - invariant under endianness changes
   - change significantly under endianness changes

Message layout would then become:
```
prefix
  pkt_length
header
  msg_type
  comm_type
  reply_type
body
  sync_field0
  ..
  sync_field1
```

Deserialisation state machines could then look for `sync_field0` at 12 bytes past the start of the `header`, and for `sync_field1` at `pkt_length` bytes after start of the `header`.

gavanderhoorn, 2016-01-25:
It might also make sense to add a `version` field in the same way. I know the protocol definition explicitly states that ensuring compatibility is up to the (driver) author, but with the expanding use of the protocol (and the 'dialects' that start to pop-up), I believe it to become really necessary to be able to discriminate between versions.

We could introduce something like an *extended header*, which basically overlays the current one.

gavanderhoorn, 2021-06-23:
We're not going to be addressing this in `simple_message` any more.

closed 2021-06-23 by gavanderhoorn