ros-industrial/industrial_core #145: robot_interface_simulator.launch doesn't find industrial_robot_simulator node

cschindlbeck, 2016-06-20:
I get the following error for 

`$ roslaunch industrial_robot_simulator robot_interface_simulator.launch `

`ERROR: cannot launch node of type [industrial_robot_simulator/industrial_robot_simulator]: can't locate node [industrial_robot_simulator] in package [industrial_robot_simulator]`

VictorLamoine, 2016-06-20:
You are probably using the package compiled from sources; make sure you have sourced the `setup.bash` file:

```
source $HOME/my_catkin_workspace/devel/setup.bash
```

cschindlbeck, 2016-06-20:
i have done that, still not working

gavanderhoorn, 2016-06-20:
From your reply to @VictorLamoine's question, I'm guessing you have a checkout of `industrial_core` in your catkin workspace? Have you built the workspace after cloning the package? Are you using regular `catkin_make`, or `catkin_tools`?

In general when reporting these kinds of problems (which is great, thanks for that), it's good to include as much information as possible. Especially build environment and steps to reproduce.

cschindlbeck, 2016-06-20:
I use ubuntu 14.04 with ROS-indigo, industrial_core is built from source with catkin_make (no errors)
git branch is indigo-devel with "Already up-to-date"
Folder structure

├── catkin_ws
|   ├── devel
|   ├── build
|   └── src
|    |   └── industrial_core
|    |   └── my_other_pkges

gavanderhoorn, 2016-06-20:
Just did the following on a clean Trusty docker container:

```shell
# usual ROS apt sources, keys, etc
$ sudo apt-get install ros-indigo-ros-base

$ mkdir -p ~/catkin_ws/src
$ cd ~/catkin_ws/src

$ git clone https://github.com/ros-industrial/industrial_core.git
$ rosdep update
$ rosdep install --from-paths ~/catkin_ws/src --ignore-src

$ cd ~/catkin_ws
$ catkin_make

$ source ~/catkin_ws/devel/setup.bash
```

Finally:
```shell
$ roslaunch industrial_robot_simulator robot_interface_simulator.launch
... logging to /home/user/.ros/log/1da8780c-36ce-11e6-a455-180373234072/roslaunch-machine-12285.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://machine:48348/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    industrial_robot_simulator (industrial_robot_simulator/industrial_robot_simulator)
    joint_trajectory_action (industrial_robot_client/joint_trajectory_action)

auto-starting new master
process[master]: started with pid [12296]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1da8780c-36ce-11e6-a455-180373234072
process[rosout-1]: started with pid [12309]
started core service [/rosout]
process[industrial_robot_simulator-2]: started with pid [12312]
process[joint_trajectory_action-3]: started with pid [12313]
^C[joint_trajectory_action-3] killing on exit
[industrial_robot_simulator-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

It seems this is working for me.

Can you list the exact steps you take to reproduce what you are seeing?

gavanderhoorn, 2016-06-20:
> i have included
```
source /opt/ros/indigo/setup.bash
source /home/my_name/catkin_ws/devel/setup.bash
```
in the bashrc file

Ignoring the double `source` there, lines in your `.bashrc` are only executed when you start a new instance of `bash` (fi by starting a new terminal, or opening a new tab), or when you explicitly `source ~/.bashrc`. If you didn't do either of those after building your workspace, you could run into the problem you reported.

cschindlbeck, 2016-06-20:
I have installed 
`sudo apt-get install ros-indigo-desktop-full`

Might this cause the problem?

Error massage 

`[rosrun] Couldn't find executable named industrial_robot_simulator below /home/my_name/catkin_ws/src/industrial_core/industrial_robot_simulator
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /home/my_name/catkin_ws/src/industrial_core/industrial_robot_simulator
[rosrun]   /home/my_name/catkin_ws/src/industrial_core/industrial_robot_simulator/industrial_robot_simulator
`

gavanderhoorn, 2016-06-20:
Also (but maybe unnecessary): if you can, please just use the released, binary distribution of `industrial_core`: those are known to work, and avoid all kinds of problems wrt dependencies and workspace layouts.

cschindlbeck, 2016-06-20:
chmod +x industrial_robot_simulator

gavanderhoorn, 2016-06-20:
> I have installed `sudo apt-get install ros-indigo-desktop-full`

> Might this cause the problem?

No. It's probably a build thing.

> I get the following error massage after `$ rosrun industrial_robot_simulator industrial_robot_simulator`
> ...

How did you clone the repository?

What is the output of:
```bash
ls -al /home/my_name/catkin_ws/src/industrial_core/industrial_robot_simulator/industrial_robot_simulator
```

gavanderhoorn, 2016-06-20:
> chmod +x industrial_robot_simulator 

The fact that you need to do that points to problems with how you downloaded the sources. It'd be nice if you could tell us *exactly* what you did.

cschindlbeck, 2016-06-20:
if i recall corectly (it has been a while actually)

```
cd catkin_ws/src
git clone https://github.com/ros-industrial/industrial_core.git
cd ../
catkin_make
```

Will i run into more of these problems with rights?

gavanderhoorn, 2016-06-20:
Is your `catkin_ws` on a non-Linux fs by any chance? Are you syncing dirs/files using dropbox/megasync/something else?

`git clone ..` should preserve permissions, and the files in the repository contain the proper permissions, so normally this should not be a problem.

cschindlbeck, 2016-06-20:
Actually i use a vanilla Ubuntu 14.04.4, no dropbox or anything related to windows.
I deleted the folder and cloned again, now it works, i probably messed up the initial clone process

thanks!

gavanderhoorn, 2016-06-20:
Ok. Good to hear it works for you as well now.

closed 2016-06-20 by gavanderhoorn