ros-industrial/industrial_core #149: skipped tests make build status: unstable

gavanderhoorn, 2016-09-05:
See [build 4](http://build.ros.org/job/Idev__industrial_core__ubuntu_trusty_amd64/4/testReport/) fi, but the general test result trend is also [unstable](http://build.ros.org/job/Idev__industrial_core__ubuntu_trusty_amd64/).

gavanderhoorn, 2016-09-05:
@shaun-edwards: I'm assuming you disabled those tests? Is there anything we can fix, so we can re-enable them?

shaun-edwards, 2016-12-22:
The tests are disabled because they need roscore to execute.  The "right" thing to do create a "rostest" to run.  Now that I've done this a few times, it should be easy.

shaun-edwards, 2017-01-16:
Back to stable: http://build.ros.org/job/Jdev__industrial_core__ubuntu_trusty_amd64/6/

gomezc, 2018-07-11:
I'll be looking at this for wrid18

gavanderhoorn, 2018-07-29:
@gomezc: were you able to make any progress with this?

gomezc, 2018-07-29:
I made some progress but was not able to complete it and have been distracted since. If no one else is going to take it, I will still work on it in my spare time, but if someone else is looking at it, I will gladly relinquish. :)

gavanderhoorn, 2018-07-29:
Could you at least describe what you did / what you found out? If/when others take this up, that may save them time.

gavanderhoorn, 2019-06-28:
It would still be nice if we could get a summary of what progress was made, but seeing as it's already been a year, that'd probably be difficult.