ros-industrial/industrial_core #168: Cannot build kinetic-devel with catkin_make

VictorLamoine, 2017-01-16:
```
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)
```

gavanderhoorn, 2017-01-16:
That is interesting. Could you try and figure out which package it is considering a 'non-catkin' package?

gavanderhoorn, 2017-01-16:
You can just `touch pkg/CATKIN_IGNORE` for that.

VictorLamoine, 2017-01-16:
Looks like it is the metapackage `industrial_core`

gavanderhoorn, 2017-01-16:
Probably [this](https://github.com/ros-industrial/industrial_core/blob/ddc74de4a12df1cf53dd137f2f88cbb013339ed9/industrial_core/CMakeLists.txt#L1) line:

```
cmake_minimum_required(VERSION 2.8.12)
```

Changed in ddc74de4a12df1cf53dd137f2f88cbb013339ed9 as part of #161.

`catkin_make` expects it to have the `VERSION` arg set to `2.8.3`.

VictorLamoine, 2017-01-16:
The problem is because we changed the cmake minimum version. For some reason catkin_make doesn't like that we specify 3.8.12, if I go back to 2.8.3 everything is ok

gavanderhoorn, 2017-01-16:
There is no need to set it to `2.8.12` here anyway, so I suggest you change it back.

gavanderhoorn, 2017-01-16:
If you can prepare a PR, I'll merge it.

VictorLamoine, 2017-01-16:
I did that to be homogeneous in the CMake versions. It's weird that it checks for a specific version. It should be "at least 2.8.3".

I'll make a merge request right now.

gavanderhoorn, 2017-01-16:
@VictorLamoine wrote:
> I did that to be homogeneous in the CMake versions. It's weird that it checks for a specific version. It should be "at least 2.8.3".

`catkin_make` does a content-check of the `CMakeLists.txt` in a metapackage, it does not parse it. There is no CMake involved at this point.

VictorLamoine, 2017-01-16:
catkin build does not complain about that.

gavanderhoorn, 2017-01-16:
No, because it's a completely different tool, and because it is capable of building mixed workspaces.

PR #169 (2017-01-16): Fix catkin_make on kinetic-devel
Fixes #168
closed 2017-01-16 by gavanderhoorn