ros-industrial/industrial_core #172: Error when building with catkin_make (for kinetic-devel)

glebysg, 2017-02-13:
When I run catkin_make I get the following error

`[  0%] Built target trajectory_msgs_generate_messages_lisp
[  0%] Built target _industrial_msgs_generate_messages_check_deps_SetRemoteLoggerLevel
[  0%] Built target _industrial_msgs_generate_messages_check_deps_ServiceReturnCode
[  0%] Built target _industrial_msgs_generate_messages_check_deps_CmdJointTrajectory
[  0%] Built target _industrial_msgs_generate_messages_check_deps_RobotStatus
[  0%] Built target _industrial_msgs_generate_messages_check_deps_SetDrivePower
[  0%] Built target _industrial_msgs_generate_messages_check_deps_StartMotion
[  0%] Built target _industrial_msgs_generate_messages_check_deps_GetRobotInfo
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target trajectory_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target _industrial_msgs_generate_messages_check_deps_DeviceInfo
[  0%] Built target _industrial_msgs_generate_messages_check_deps_DebugLevel
[  0%] Built target trajectory_msgs_generate_messages_py
[  0%] Built target _industrial_msgs_generate_messages_check_deps_RobotMode
[  0%] Built target _industrial_msgs_generate_messages_check_deps_TriState
[  0%] Built target _industrial_msgs_generate_messages_check_deps_StopMotion
[  0%] Built target trajectory_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target trajectory_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_py
[  2%] Built target simple_message_dummy
[  2%] Built target geometry_msgs_generate_messages_eus
[  4%] Built target industrial_robot_client_dummy
[ 10%] Built target industrial_msgs_generate_messages_lisp
[ 17%] Built target industrial_msgs_generate_messages_eus
[ 18%] Built target industrial_utils
[ 26%] Built target industrial_msgs_generate_messages_py
[ 32%] Built target industrial_msgs_generate_messages_cpp
[ 39%] Built target industrial_msgs_generate_messages_nodejs
[ 39%] Building CXX object industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/n_point_filter.cpp.o
[ 39%] Building CXX object industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/add_smoothing_filter.cpp.o
[ 40%] Building CXX object industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/uniform_sample_filter.cpp.o
[ 41%] Building CXX object industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/smoothing_trajectory_filter.cpp.o
[ 41%] Built target industrial_msgs_generate_messages
[ 70%] Built target simple_message
[ 70%] Built target simple_message_float64
[ 83%] Built target simple_message_bswap
[ 86%] Built target industrial_robot_client
[ 91%] Built target industrial_robot_client_bswap
[ 92%] Built target motion_streaming_interface
[ 93%] Built target motion_download_interface
[ 94%] Built target robot_state
[ 96%] Built target joint_trajectory_action
[ 97%] Built target motion_streaming_interface_bswap
[ 98%] Built target motion_download_interface_bswap
[100%] Built target robot_state_bswap
In file included from /opt/ros/kinetic/include/moveit/robot_state/robot_state.h:41:0,
                 from /opt/ros/kinetic/include/moveit/robot_trajectory/robot_trajectory.h:41,
                 from /opt/ros/kinetic/include/moveit/planning_interface/planning_response.h:40,
                 from /opt/ros/kinetic/include/moveit/planning_interface/planning_interface.h:42,
                 from /opt/ros/kinetic/include/moveit/planning_request_adapter/planning_request_adapter.h:41,
                 from /home/glebys/catkin_ws/src/industrial_core/industrial_trajectory_filters/include/industrial_trajectory_filters/filter_base.h:39,
                 from /home/glebys/catkin_ws/src/industrial_core/industrial_trajectory_filters/include/industrial_trajectory_filters/uniform_sample_filter.h:35,
                 from /home/glebys/catkin_ws/src/industrial_core/industrial_trajectory_filters/src/uniform_sample_filter.cpp:32:
/opt/ros/kinetic/include/moveit/robot_model/robot_model.h:72:26: error: ‘ModelInterfaceSharedPtr’ in namespace ‘urdf’ does not name a type
   RobotModel(const urdf::ModelInterfaceSharedPtr &urdf_model, const srdf::ModelConstSharedPtr &srdf_model);
                          ^
/opt/ros/kinetic/include/moveit/robot_model/robot_model.h:99:15: error: ‘ModelInterfaceSharedPtr’ in namespace ‘urdf’ does not name a type
   const urdf::ModelInterfaceSharedPtr &getURDF() const
               ^
/opt/ros/kinetic/include/moveit/robot_model/robot_model.h:446:9: error: ‘ModelInterfaceSharedPtr’ in namespace ‘urdf’ does not name a type
   urdf::ModelInterfaceSharedPtr urdf_;
[... 64 lines cut ...]
   urdf::ModelInterfaceSharedPtr urdf_;
         ^
In file included from /opt/ros/kinetic/include/moveit/planning_request_adapter/planning_request_adapter.h:42:0,
                 from /home/glebys/catkin_ws/src/industrial_core/industrial_trajectory_filters/src/add_smoothing_filter.cpp:37:
/opt/ros/kinetic/include/moveit/planning_scene/planning_scene.h:96:29: error: ‘ModelInterfaceSharedPtr’ in namespace ‘urdf’ does not name a type
   PlanningScene(const urdf::ModelInterfaceSharedPtr &urdf_model, const srdf::ModelConstSharedPtr &srdf_model,
                             ^
/opt/ros/kinetic/include/moveit/planning_scene/planning_scene.h:938:66: error: ‘ModelInterfaceSharedPtr’ in namespace ‘urdf’ does not name a type
   static robot_model::RobotModelPtr createRobotModel(const urdf::ModelInterfaceSharedPtr &urdf_model,
                                                                  ^
industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/build.make:86: recipe for target 'industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/uniform_sample_filter.cpp.o' failed
make[2]: *** [industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/uniform_sample_filter.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/build.make:62: recipe for target 'industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/n_point_filter.cpp.o' failed
make[2]: *** [industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/n_point_filter.cpp.o] Error 1
industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/build.make:110: recipe for target 'industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/add_smoothing_filter.cpp.o' failed
make[2]: *** [industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/add_smoothing_filter.cpp.o] Error 1
industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/build.make:134: recipe for target 'industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/smoothing_trajectory_filter.cpp.o' failed
make[2]: *** [industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/src/smoothing_trajectory_filter.cpp.o] Error 1
CMakeFiles/Makefile2:4865: recipe for target 'industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/all' failed
make[1]: *** [industrial_core/industrial_trajectory_filters/CMakeFiles/industrial_trajectory_filters.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`

shaun-edwards, 2017-02-13:
See if PR #170 fixes things.  I'll merge the PR just as soon as the Travis builds complete (presumably successful).

gavanderhoorn, 2017-10-07:
I think this was fixed with the merge of #170.

I just compiled all `industrial_core` packages from source on Lunar and did not run into this issue any more.

gavanderhoorn, 2017-10-07:
@glebysg: please re-open if you feel this has not been addressed.

And thanks for reporting.

closed 2017-10-07 by gavanderhoorn