ros-industrial/industrial_core #173: Tests fails with catkin build on ROS Kinetic

VictorLamoine, 2017-02-16:
Create a workspace:
```bash
mkdir -p catkin_workspace/src
cd catkin_workspace/src
git clone https://github.com/ros-industrial/industrial_core.git
cd ..
```

Run tests with catkin tools:
```bash
catkin run_tests --no-status --force-color
```

Errors:
```bash
make[3]: Leaving directory '/home/victor/catkin_workspace/build/simple_message'
[ 84%] Built target gtest_main
/home/victor/catkin_workspace/src/industrial_core/simple_message/src/robot_status.cpp:44:39: fatal error: industrial_msgs/RobotMode.h: No such file or directory

                                       ^
compilation terminated.
CMakeFiles/simple_message_float64.dir/build.make:518: recipe for target 'CMakeFiles/simple_message_float64.dir/src/robot_status.cpp.o' failed
make[3]: *** [CMakeFiles/simple_message_float64.dir/src/robot_status.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/home/victor/catkin_workspace/src/industrial_core/simple_message/src/robot_status.cpp:44:39: fatal error: industrial_msgs/RobotMode.h: No such file or directory

                                       ^
compilation terminated.
CMakeFiles/simple_message.dir/build.make:518: recipe for target 'CMakeFiles/simple_message.dir/src/robot_status.cpp.o' failed
make[3]: Leaving directory '/home/victor/catkin_workspace/build/simple_message'
make[3]: *** [CMakeFiles/simple_message.dir/src/robot_status.cpp.o] Error 1
make[2]: *** [CMakeFiles/simple_message.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1668: recipe for target 'CMakeFiles/simple_message.dir/all' failed
/home/victor/catkin_workspace/src/industrial_core/simple_message/src/robot_status.cpp:44:39: fatal error: industrial_msgs/RobotMode.h: No such file or directory

                                       ^
compilation terminated.
CMakeFiles/simple_message_bswap.dir/build.make:518: recipe for target 'CMakeFiles/simple_message_bswap.dir/src/robot_status.cpp.o' failed
make[3]: Leaving directory '/home/victor/catkin_workspace/build/simple_message'
make[3]: *** [CMakeFiles/simple_message_bswap.dir/src/robot_status.cpp.o] Error 1
make[2]: *** [CMakeFiles/simple_message_bswap.dir/all] Error 2
CMakeFiles/Makefile2:1924: recipe for target 'CMakeFiles/simple_message_bswap.dir/all' failed
make[3]: Leaving directory '/home/victor/catkin_workspace/build/simple_message'
```

VictorLamoine, 2017-02-16:
Seems like it is just a build order problem. Message is being generated after trying to compile `robot_status.cpp`.

If I run `catkin build` and the run the tests it goes fine.

gavanderhoorn, 2017-02-16:
In any case this should not happen.

Perhaps something changed in Kinetic, or `catkin_tools`. [simple_message/CMakeLists.txt](https://github.com/ros-industrial/industrial_core/blob/ddc74de4a12df1cf53dd137f2f88cbb013339ed9/simple_message/CMakeLists.txt#L102) does declare the dependency on `industrial_msgs`.

VictorLamoine, 2017-02-16:
I can't find what's wrong.

```
$ catkin --version
catkin_tools 0.4.3 (C) 2014-2017 Open Source Robotics Foundation
catkin_tools is released under the Apache License, Version 2.0 (http://www.apache.org/licenses/LICENSE-2.0)
---
Using Python 2.7.12 (default, Nov 19 2016, 06:48:10) [GCC 5.4.0 20160609]

```

gavanderhoorn, 2017-10-07:
@VictorLamoine: I'm going to chalk this one up to `catkin_tools` doing weird things sometime and close the issue.

`industrial_msgs_EXPORTED_TARGETS` does actually contain all the necessary targets:
```
industrial_msgs_generate_messages_cpp;industrial_msgs_generate_messages_lisp;industrial_msgs_generate_messages_py;std_msgs_generate_messages_cpp;std_msgs_generate_messages_lisp;std_msgs_generate_messages_py;trajectory_msgs_generate_messages_cpp;trajectory_msgs_generate_messages_lisp;trajectory_msgs_generate_messages_py;std_msgs_generate_messages_cpp;std_msgs_generate_messages_lisp;std_msgs_generate_messages_py;geometry_msgs_generate_messages_cpp;geometry_msgs_generate_messages_lisp;geometry_msgs_generate_messages_py;std_msgs_generate_messages_cpp;std_msgs_generate_messages_lisp;std_msgs_generate_messages_py
```

I agree with you that using `run_tests` directly should result in `catkin` first building all pkgs in the correct order though.

It would seem this is a `catkin_tools` issue and not with this package.

closed 2017-10-07 by gavanderhoorn

VictorLamoine, 2017-10-09:
I tested again and I cannot reproduce the issue so it's fine that you closed it.