ros-industrial/industrial_core #175: robot_client: JointTrajectoryStreamer does not send TRAJ_STOP if in IDLE state

gavanderhoorn, 2017-03-29:
As per subject really.

If `JointTrajectoryStreamer::jointTrajectoryCB(..)` receives an empty `JointTrajectory` msg while the streamer state machine is in the `IDLE` state, it does not call `trajectoryStop()`, resulting in no `STOP_TRAJECTORY` message to the trajectory server (see [here](https://github.com/ros-industrial/industrial_core/blob/d18b676f30616bb29b92dede04120b364907f8c7/industrial_robot_client/src/joint_trajectory_streamer.cpp#L74-L85)). Sending an empty trajectory is used by `JointTrajectoryAction::cancelCB(..)` to stop execution of the currently executing trajectory ([here](https://github.com/ros-industrial/industrial_core/blob/aca48389468a53ca932373b8f94f0b3ea1bdc9db/industrial_robot_client/src/joint_trajectory_action.cpp#L183-L201)).

For drivers with points buffers on the controller the streamer could already be in `IDLE` while the robot is still executing the trajectory (as the last `N` points of the traj have been buffered). This leads to the `STOP` never being transmitted and thus the cancellation of goals potentially being ignored.

gavanderhoorn, 2017-03-29:
The question is thus: is this desired behaviour? Can any of the authors (@shaun-edwards, @JeremyZoss) perhaps comment on what the rationale was to ignore empty trajectories in `IDLE`?

Should empty messages not be used to cancel goals?

Or was the "point buffer on server" case not considered when this code was written?

JeremyZoss, 2017-03-29:
+1

I think you are correct, that this case (buffered points on controller) was not considered.  It's a bit difficult to review the history, since Google Code is mostly shutdown.  But it looks like `industrial_robot_client` mostly replicated the existing behavior of the original motoman `joint_trajectory_streamer` code.

I think an easy patch would be to:
  - revise the log message (line [89](https://github.com/ros-industrial/industrial_core/blob/d18b676f30616bb29b92dede04120b364907f8c7/industrial_robot_client/src/joint_trajectory_streamer.cpp#L89))
 - add a `trajectory_stop` block just below that message, similar to what's currently at lines [81-83](https://github.com/ros-industrial/industrial_core/blob/d18b676f30616bb29b92dede04120b364907f8c7/industrial_robot_client/src/joint_trajectory_streamer.cpp#L81-L83)

Alternatively, you could probably spend a bit more time cleaning things up make the "stop" conditions a bit clearer.

gavanderhoorn, 2021-06-21:
#260 should fix this.

closed 2021-06-21 by gavanderhoorn