ros-industrial/industrial_core #176: Implementation of simple_message_hw_interface

rsinnet, 2017-05-05:
Currently, ros-industrial does _not_ have a `simple_message_hw_interface` implementation using `ros_control`. I would like to implement such an interface so I can take advantage of `ros_control` features. I'm particularly interested in trajectory replacement. The use case is something like:
1. Send a trajectory starting at time _t_0_ ( < _t_1_ < _t_2_) that finishes at _t_2
2. Send a new trajectory before _t_1_ that starts _t_1_.
In this scenario I would like to see blending of the old trajectory to the new trajectory at time _t_1_ as described in: [Understanding Trajectory Replacement](http://wiki.ros.org/joint_trajectory_controller/UnderstandingTrajectoryReplacement)

@gavanderhoorn Any thoughts here would be very helpful. Thanks!

gavanderhoorn, 2017-05-05:
Some quick thoughts:

 - a single hw interface will have to combine the `robot_state` and `trajectory_streamer` nodes in a meaningful way (two sockets, non-blocking)
 - the hw interface will fall in the *streaming* driver category (this would make it incompatible with drivers that only support downloading)
 - I'm not sure how we should treat sequence nrs: officially they are to start at `0`, increase with each new traj pt and only ever restart at `0` when a new traj starts. With traj replacement, the chances of starting at the first pt are rather low. I'm not certain how existing drivers will respond to that.
 - it would be a good idea to add support for `JOINT_TRAJ_PT_FULL` from the start. That way things like acceleration and effort fields would be supported (although there are not many server implementations that will actually make use of those)
 - it could even be a good idea to start working on an [IRCv2](https://github.com/ros-industrial/industrial_experimental/tree/indigo-devel/IRC_v2) implementation (but we don't really have any server implementations that would be compatible with that)
 - I expect that for certain servers (fi the one in `fanuc_driver` and `fanuc_driver_exp`), maximum attainable control rate will be rather low, but that is something that can be configured on the `ros_control` side
 - some server implementations (fi `motoros`) have an implicit assumption that they are fed complete trajectories. I'm guessing an 'infinite' `ros_control` trajectory could work (see 'sequence nrs' bullet), but we'll have to see

gavanderhoorn, 2017-05-05:
> I'm particularly interested in trajectory replacement. The use case is something like:

this will obviously all come 'for free' once a suitable hw interface has been implemented.

Only real requirement for this is that the hw interface implement a *streaming* approach. Trajectory replacement with a downloading driver could be possible, but would really depend on controller-local implementation details which I don't think we can make any assumptions about in the hw interface.

rsinnet, 2017-05-05:
Thanks! This should really help.

gavanderhoorn, 2017-05-05:
Know btw that `fanuc_driver` and `fanuc_driver_exp` currently only support `JOINT_TRAJ_PT` (for enqueuing new points) and `JOINT_POSITION` (for reporting joint state).

I could easily add support for / switch `fanuc_driver_exp` to `JOINT_TRAJ_PT_FULL`, but that would be a bit pointless, as on Fanuc controllers we can't control velocity, acceleration or effort in any meaningful way anyway.

gavanderhoorn, 2017-05-05:
> a single hw interface will have to combine the `robot_state` and `trajectory_streamer` nodes in a meaningful way (two sockets, non-blocking)

btw: I'm not implying with this bullet that any hw interface implementation should reimplement the network protocol (de)serialisation. If at all possible, reuse what's provided in `simple_message` and `industrial_robot_client`.

rsinnet, 2017-05-05:
Thanks, I will try to do a minimal implementation as you suggest.

gavanderhoorn, 2017-05-05:
O and if you are targeting Fanuc: please develop against [fanuc_driver_exp](https://github.com/gavanderhoorn/fanuc_driver_exp).

It's going to replace `fanuc_driver`, and should prove much more amenable to the kind of interaction that you are after.

rsinnet, 2017-05-05:
Thanks, I've been using [fanuc_driver_exp](https://github.com/gavanderhoorn/fanuc_driver_exp) since it has support for the robot I'm using.

gavanderhoorn, 2017-05-05:
Both drivers should be manipulator agnostic, so I'm slightly confused by your statement.

rsinnet, 2017-05-05:
Sorry, I confused [fanuc_driver_exp](https://github.com/gavanderhoorn/fanuc_driver_exp) with [fanuc_experimental](https://github.com/ros-industrial/fanuc_experimental).

gavanderhoorn, 2017-06-02:
@rsinnet: have you had any success? Not checking up on you, just curious.

rsinnet, 2017-06-03:
@gavanderhoorn Thanks for your interest. I haven't been able to obtain consistent performance with the ROS-I implementation. I'm trying to follow trajectories that are specified in terms of desired positions and velocities as closely as possible. I've been speaking regularly with FANUC engineers and management and they have mentioned that streaming joint positions into `PR` registers may not be consistent because they prevent lookahead which is something the FANUC motion planning system does―there is currently no way to disable it―whereas using positions specified in `P` registers (hard-coded in a TP file) allows the interpreter to look ahead and blend trajectories (something that a roboticist wouldn't generally want but better than inconsistency IMO).

To test this, I wrote a driver stack that effectively generates and sends TP programs and then executes them in a similar fashion to the ROS-I KAREL code―I'm glancing over the details here but you can probably guess what was needed like buffering of TP programs, TCP packets, FTP transfer of TP programs, etc. This approach has produced more consistent results. Because there is some look-ahead, this approach certainly has latency which makes it impractical for feedback control (depending on the timescale). That being said, your addition of the buffer in fanuc_driver_exp to handle network latency (as I believe I noticed in a comment or README) leads to this same foregone conclusion. I haven't thoroughly examined all the timings yet, but it may be possible to improve what I've written to achieve more consistent behavior even if it is blended by putting the points in a TP program as registers, but the required buffer size and thus latency may be large.

They also suggested that 40 ms (5 INT periods at 8ms setting) is the minimum practical time between points. They say that it depends on what is running in the background. I'm not convinced but I'll let you make your own judgment on that. Another point their head of engineering pushed was that, instead of sending the J command, it would be more effective to send the L command and specify the velocity as the speed in mm/sec of the TCP. This requires setting the TCP on the FANUC controller (\*sigh\*). I have played around with this and can share some early results but I haven't really finished analysis here. At any rate, I'm currently tackling these issues and making sure I can get close enough to desired performance before I go any further.

If I am able to get to consistent performance with the FANUC and it makes sense, I will continue and implement a `ros_control` interface.

gavanderhoorn, 2017-06-04:
Thanks. You've basically confirmed everything I've also already found in the past few years - unfortunately.

re: generating TP programs & look ahead: yes, I've played with that as well - basically turning the streaming driver into a downloading one - but wasn't happy with that either. If only for the fact that it puts an upper limit on the length of trajectories.

re: smoothness vs accuracy: *this* is the biggest problem with anything that makes use of what Fanuc currently has. The necessity for the controller to maintain continuity of the motion - at least when interfaced through the regular channels - makes any sort of meaningful external control essentially impossible. Point-to-point is definitely possible, but continuous motion seems undoable.

re: 40 ms: I can understand why they'd say that.

re: linear motions: yeah. That might be, but the problem is we're going to run into all sorts of nasty-ness with singulaties, arm configurations and turn numbers. I'm definitely interested in adding a Cartesian interface to the driver, but probably not as the default one.

As to setting the TCP: I would think you'd be able to transform whatever Cartesian pose would be specified as the waypoint back to `tool0` (in the urdf), which should allow you to get away with just always assuming an all-zeros toolframe configuration on the controller.

---

Overall I believe it would be worthwhile for Fanuc to finally start implementing an external motion control interface similar to what almost all their competitors have. Not just for ROS-Industrial, but just to make applications possible that require more than simple point-to-point and / or linearly interpolated motion between two points.

Having to chose between CNT100 and thus smoothness or a lower CNT and thus accuracy is not a viable solution.

gavanderhoorn, 2017-06-04:
> Another point their head of engineering pushed was that, instead of sending the J command, it would be more effective to send the L command and specify the velocity as the speed in mm/sec of the [...]

I would be interested in those results. I've spend quite some time on many different approaches but in the end never released / finished (m)any of them because I was not happy with the results. I have two / three more tricks I want to try before giving up and waiting for a better motion interface.

If it's more convenient, you can find my email address in any commit to this repository.

gavanderhoorn, 2017-06-04:
And a finaly comment - seeing as this is a ticket on `industrial_core`: note that the issues you are having are a result of the characteristics / shortcomings of the Fanuc platform, not necessarily of using `simple_message` or the `industrial_robot_client` infrastructure.

Not claiming either of those is perfect, but it's the way received traj pts are communicated to the motion side of the Fanuc platform that limits us here.

gavanderhoorn, 2017-06-30:
@rsinnet: curious again: any progress?

wwerst, 2018-05-17:
@gavanderhoorn I recommend you look at using the HMI option (R553) for streaming points to the robot instead of Karel. I am using HMI to write to `PR[1]@27.19` and get robot position feedback from `POS[0]@27.19`, and it is working well. Then, for the TP program to actually go to the position register repeatedly, it is as simple as:
```
1:  LBL[1] ;
2:J PR[1] 100% CNT100    ;
3:  JMP LBL[1] ;
```

I'll put this as an issue in gavanderhoorn/fanuc_driver_exp and link it back here. The code I have isn't ready for release yet, but I figured I'd give you a heads up about a better approach for streaming joint positions to the robot.

gavanderhoorn, 2021-06-22:
While this would be great, we're not going to implement a `ros_control` compatible `hardware_interface` for `simple_message` and/or `industrial_robot_client` here.

The world has changed, and simple communication protocols are now much more widely available than when `simple_message` was implemented. Maintaining our own implementation of such a protocol incurs quite some maintenance overhead, so ideally we'd use whatever exists.

Whether that would be in combination with `ros_control`/`ros2_control` or not would be up to the developer of such an integration.

closed 2021-06-22 by gavanderhoorn