ros-industrial/industrial_core #177: tcp server makeConnect() blocks forever

jrgnicho, 2017-07-07:
In the absence of a connection, the `TcpServer::makeConnect()` method blocks the main thread and it won't exit even when  the program is terminated with Ctrl-C.  I can get around this by placing calling makeConnect() from a second thread however this alternative feels unsafe as the an unhandled exception is thrown when the program exits.
I ran my node in ubuntu 16.04 with ros-kinetic.

gavanderhoorn, 2017-07-07:
One way to fix this is to put [this line](https://github.com/ros-industrial/industrial_core/blob/12a74a1f9f26aea0ee075edaf4c84473bd8e112a/simple_message/src/socket/tcp_server.cpp#L135) in a `while` with a `select(..)` and only `accept(..)` whenever `select(..)` returns instead of timing out.

jrgnicho, 2017-07-07:
@gavanderhoorn  I tried using the `select(...)` approach you had suggested but it would error out even when a client had already connected, I followed the manual [here ](http://beej.us/guide/bgnet/output/html/multipage/advanced.html#select) for guidance on the proper use of `select` 

What worked for me was to add the following lines around the `accept()` call 
```
    // disable blocking
    int opts = fcntl(this->getSrvrHandle(),F_GETFL,0);
    opts |= O_NONBLOCK;
    fcntl(this->getSrvrHandle(), F_SETFL, opts);

    // accepting connection
    rc = ACCEPT(this->getSrvrHandle(), NULL, NULL);

    // re-enable blocking
    opts &= ~O_NONBLOCK;
    fcnlt(this->getSrvrHandle(), F_SETFL, opts);
```

If this solution is desirable I could submit a PR however some of the unit tests did fail after making this change.

shaun-edwards, 2017-07-08:
@jrgnicho, this goes beyond my knowledge of the socket library.  How did you come up with this solution?  Is there a reference somewhere?

Can you provide a test for this?  I'm not entirely sure how you issue a `Ctrl+C` programmatically, but it might be possible.

jrgnicho, 2017-07-09:
Here is a [link ](http://www.beej.us/guide/bgnet/output/html/multipage/advanced.html#blocking) that describes how to disable blocking.  However, I decided not to use that approach since it caused some of the unit tests in simple message to [fail](https://github.com/jrgnicho/industrial_core/tree/fix-tcp-server-connect-blocking).
My final solution was to call makeConnect() from a second thread and then call [shutdown](https://stackoverflow.com/questions/18267146/how-to-let-a-thread-which-blocks-on-recv-exit-gracefully) on the socket handle from the main thread.  Perhaps we can add a ```close``` method to TcpServer that just calls ```shutdown``` on the server handle.