ros-industrial/industrial_core #178: How to create user-defined message

Booooooosh, 2017-08-04:
Apart from JointTrajectory message, I want extra information(Typically booleans and ints) to be sent to the robot controller. Is there any tutorials on how to create a user-defined message? I have a vague idea of what I need to do, like, defining my own message, send it through the socket and decoding on the robot controller side, but not quite sure how to do this.

Many thanks!

shaun-edwards, 2017-08-05:
This deserves a more complete answer.  I hope to provide one in a couple of days.  If you need some immediate help, this [wiki](http://wiki.ros.org/simple_message) might be helpful.

Booooooosh, 2017-08-05:
Thanks for the fast reply. Actually I have been looking at the Wiki page, but a tutorial would be great! Looking forward to it!

gavanderhoorn, 2017-08-10:
Not a complete answer, but perhaps something that can function as an example: see ros-industrial/motoman#93 where I did the same, but for the `motoman` driver specifically. The idea is the same for `industrial_core` though.

Booooooosh, 2017-08-17:
@gavanderhoorn Sorry for the delayed reply. Many thanks! I will look into the industrial-core to see whether I can do the same for Fanuc driver.

gavanderhoorn, 2017-10-07:
@shaun-edwards: do you see any chance to create that wiki page that you mentioned?

Otherwise I suggest we either close this issue and open a new one that specifically asks for documentation on creating custom messages, or we update the title of this one to reflect the need for documentation

shaun-edwards, 2017-10-07:
After further review, I believe the documentation is there.  A tutorial would help with the specific task of creating a new message type, but it's not high on the priority list.

gavanderhoorn, 2017-10-07:
Could you post a link to the docs you are referring to?

shaun-edwards, 2017-10-07:
They are linked above and in the API documentation.  There could always be more, but I do not plan to create a specific tutorial.

gavanderhoorn, 2017-10-07:
I think I'm looking in the wrong places: do you mean the `simple_message` wiki page? Do the *Code API* pages include any guidance on creating user defined messages?

shaun-edwards, 2017-10-07:
There is enough information in the wiki and code API docs that one could create a custom message. Furthermore, there are several examples of "custom" messages in simple message. While not an explicit tutorial, I believe there is enough for developers to extract the info they need. 

I agree a tutorial may help, but it is not high on the priority list given the existing sources of info.

gavanderhoorn, 2017-10-08:
Then I'll close this issue.

gavanderhoorn, 2017-10-08:
I think it's not too straightforward for new users to figure out how to add custom messages, as it's quite convoluted at the moment actually.

closed 2017-10-08 by gavanderhoorn