ros-industrial/industrial_core #180: industrial_utils: using 'default joint names' when parameter cannot be found is a bad idea

gavanderhoorn, 2017-10-02:
In `industrial_utils/param_utils::getJointNames(..)`:

https://github.com/ros-industrial/industrial_core/blob/9a0fe17aa129fe1857dccf54043ab534379eb924/industrial_utils/src/param_utils.cpp#L122-L129

If either `controller_joint_names` cannot be found, or the urdf cannot be parsed for joint names, `getJointNames(..)` returns a set of 'default names'.

This is not a good behaviour: the function should fail if it cannot find any joint names. Making some names up (even though they follow ros-i conventions) and then returning successfully makes for poor reproducibility and complicates debugging.

Ridhwanluthra, 2018-07-11:
hi, is this as straightforward as raising as error and returning?

Ridhwanluthra, 2018-07-11:
cool, fixing this as a part of #wrid18

Levi-Armstrong, 2019-02-06:
PR #195 has been merged closing.

closed 2019-02-06 by Levi-Armstrong