ros-industrial/industrial_core #210: "Trajectory splicing not yet implemented, stopping current motion"

shreyasgan, 2018-08-22:
Hello, 

I am trying to send my Staubli TX90 robot arm a Joint Trajectory command which publishes to /joint_path_command, which the motion_streaming_interface node subscribes to. My code complies and build correctly, but when I run the node, there is an error saying "Trajectory splicing not yet implemented, stopping current motion." I looked online and noticed that:
 // read current state value (should be atomic)
   int state = this->state_;

   ROS_DEBUG("Current state is: %d", state);
   if (TransferStates::IDLE != state)
  {
    if (msg->points.empty())
       ROS_INFO("Empty trajectory received, canceling current trajectory");
    else
      ROS_ERROR("Trajectory splicing not yet implemented, stopping current motion.");
 
         this->mutex_.lock();
    trajectoryStop();
         this->mutex_.unlock();
    return;

This is my code for the node that I am running to communicate with the robot arm and subscribe to /joint_states and publish to /joint_path_command:

using namespace std;

sensor_msgs::JointState JS;

void jointStateCallback(const sensor_msgs::JointState msg)
{
	JS = msg;
}

int main(int argc, char** argv)
{
	ros::init(argc, argv, "mover_node");
	ros::NodeHandle n;
	ros::AsyncSpinner spinner(1);
	spinner.start();  
	bool success;
	ros::Publisher joint_pub = n.advertise<trajectory_msgs::JointTrajectory>("/joint_path_command", 1);
	ros::Subscriber joint_sub = n.subscribe<sensor_msgs::JointState>("/joint_states", 1, jointStateCallback);	
	
	ros::Rate rate(10.0);

	trajectory_msgs::JointTrajectory JT;
	
	JT.joint_names.clear();
	JT.joint_names.push_back("joint_1");
	JT.joint_names.push_back("joint_2");
	JT.joint_names.push_back("joint_3");
	JT.joint_names.push_back("joint_4");
	JT.joint_names.push_back("joint_5");
	JT.joint_names.push_back("joint_6");

	JT.points.resize(1);
	JT.points[0].positions.resize(JT.joint_names.size());

	while(ros::ok())
{
	//if(JS.position.size() != 0)
		cout << JT << endl;
	JT.points[0].positions = JS.position;
        JT.points[0].positions += 0.1;
	
	JT.points[0].time_from_start = ros::Duration(0.0001);
	
		
	if(JS.position.size() != 0)
		joint_pub.publish(JT);
	rate.sleep();

}

gavanderhoorn, 2018-08-22:
You also posted this on ROS Answers ([here](https://answers.ros.org/question/301266/trajectory-splicing-error/)). I've responded there.

For the future: please refrain from cross-posting like this. It is not a very nice practice.

closed 2018-08-22 by gavanderhoorn

SunnyKatyara, 2019-07-28:
Dear shreyasgan, i am also facing the same issue. I have cartesian planner that gives me out on the topic with sensor_msgs/JointState and in order to command the robot, the type of topic is trajectory_msgs/TrajectoryState. Can you please help me how did you do that?

Thank You