ros-industrial/industrial_core #213: Melodic support

jorgenfb, 2018-10-16:
Currently the install instructions references melodic packages which are not available quote from http://wiki.ros.org/Industrial/Install:

```
   * Industrial core: `apt-get install ros-melodic-industrial-core`
   * ABB stack: `apt-get install ros-melodic-abb`
   * Universal robot stack: `apt-get install ros-melodic-universal-robot`
   * ros_canopen stack: `apt-get install ros-melodic-ros-canopen`
```

gavanderhoorn, 2018-10-16:
Duplicate of https://github.com/ros-industrial/ros_industrial_issues/issues/40.

gavanderhoorn, 2018-10-16:
This is an unfortunate result of using the `ROS_DISTRO` variable on that page.

See the referenced issue for more details.

closed 2018-10-16 by gavanderhoorn