ros-industrial/industrial_core #225: Incorrect return value on loading JointTrajPt?

gonzalocasas, 2019-05-05:
I am not entirely sure, but this line looks like a bug to me: https://github.com/ros-industrial/industrial_core/blob/kinetic-devel/simple_message/src/joint_traj_pt.cpp#L111

It will return `true` even if the duration failed to load on the buffer. I don't know if this is for some reason an intended behavior, but it does feel more like a copy&paste bug. In that case, the fix should probably be to flatten this very nested construct to avoid this kind of bugs from creeping in.

gavanderhoorn, 2019-05-06:
That does look like it shouldn't be there, as `rtn` is now set to `true` regardless of whether loading succeeded.

gavanderhoorn, 2019-05-06:
Impact of this is rather low btw, as `JointTrajPt::load(..)` is only used when deserialising a `JointTrajPt` from an incoming SimpleMessage connection. `JointTrajPt`s are typically only serialised by `simple_message`, as server implementations are generally written in the OEM's language.

gonzalocasas, 2019-05-06:
> as to the ladder programming: yes, I'm not a fan of that either, but "the fix" would be to remove that line.

Fair enough. I'll send a PR for the fix.

gonzalocasas, 2019-05-24:
@gavanderhoorn ping ;)

commit 75b980a8 (2019-05-24): fix #225: incorrect return value if deserializing duration fails on joint_traj_pt (#226)
closed 2019-05-24 by gavanderhoorn