ros-industrial/industrial_core #230: Joint names do not match

mjt28, 2019-06-18:
I am trying to integrate the industrial_core package with my robot moveit config to and want to try and run it with ros_descartes. However, I keep getting an error:
`Joint trajectory action failing on invalid joints`

On further investigation, I found out that my, `gh.getGoal()->trajectory.joint_names` is empty. Can someone help me with this please?

gavanderhoorn, 2019-06-18:
> On further investigation, I found out that my, `gh.getGoal()->trajectory.joint_names` is empty. Can someone help me with this please?

that would seem to be a problem with who/whatever is generating your trajectory, not in `industrial_robot_client`.

gavanderhoorn, 2019-06-18:
For now I'm going to assume this is not an issue with the packages in this repository, so I'm closing this.

Feel free to keep commenting on it.

closed 2019-06-18 by gavanderhoorn