ros-industrial/industrial_core #249: 通过Industrial_trajectory_filters输出的轨迹点，隔一秒会发生跳跃到他本该在的点的一秒以后的点之后的情况

kobesdu, 2019-12-10:
我现在已经可以用这个工具包输出时间步长相等的规划点，但是打印出来的点有些问题，他每隔大约1s的时间，一个点会跳到差不多这个点1s之后，他离开了他原本应该在的地方，而是跑到了后面的position序列中，但是他的时间戳还是原来的时间戳。
例如 1.05秒是第100个点，1.1秒本该是第101个点，但是结果却是：第101个点会是1.1秒规划的正确数值，但是在1.95秒之后，又会出现一个1.1秒的点和该点的数值。
[我某次规划中获取的点的数据（未使用此包获取的规划结果）](https://img-blog.csdnimg.cn/201912101634421.png?x-oss-process=image/watermark,type_ZmFuZ3poZW5naGVpdGk,shadow_10,text_aHR0cHM6Ly9ibG9nLmNzZG4ubmV0L2tvYmVzZHU=,size_16,color_FFFFFF,t_70)
[使用该包以后获得的结果](https://img-blog.csdnimg.cn/20191210163308926.png?x-oss-process=image/watermark,type_ZmFuZ3poZW5naGVpdGk,shadow_10,text_aHR0cHM6Ly9ibG9nLmNzZG4ubmV0L2tvYmVzZHU=,size_16,color_FFFFFF,t_70)

kobesdu, 2019-12-10:
上面的链接是我生成的数据点的曲线。因为我不知道怎么上传图片。

gavanderhoorn, 2019-12-10:
I'm sorry, but I'm going to have to request you post issues here in English.

closed 2019-12-10 by gavanderhoorn

kobesdu, 2019-12-10:
> I'm sorry, but I'm going to have to request you post issues here in English.

sorry ,my English is poor.
now i translate my post to English.
 I can use the Industrial_trajectory_filters to output planning points , but there are some problems with the printed points. 

For example, 1.05 second is the 100th point, 1.1 second should be the 101st point, but the result i get is: the 1.1 second  point is 101th point(record the point as A) ,and its positions,velocities,accelerations,effort values are right.but,the 1.1 second  point  also appear after the 1.95 second  point,and its its positions,velocities,accelerations,effort values is not the same with the A points.
please look at my links above,you may understand my means .that pictures are the value of positions-times