ros-industrial/industrial_core #252: catkin_make not working for ROS

ishaniis, 2020-02-27:
/usr/include/c++/5/bits/predefined_ops.h:194:17: note:   ‘std::__cxx11::basic_string<char>’ is not derived from ‘const std::set<_Key, _Compare, _Alloc>’
  { return *__it == _M_value; }
                 ^
In file included from /usr/include/boost/config/no_tr1/complex.hpp:21:0,
                 from /usr/include/boost/math/policies/error_handling.hpp:15,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/console.h:39,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/interface_manager.h:38,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/include/combined_robot_hw/combined_robot_hw.h:35,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:29:
/usr/include/c++/5/complex:468:5: note: candidate: template<class _Tp> bool std::operator==(const _Tp&, const std::complex<_Tp>&)
     operator==(const _Tp& __x, const complex<_Tp>& __y)
     ^
/usr/include/c++/5/complex:468:5: note:   template argument deduction/substitution failed:
In file included from /usr/include/c++/5/bits/stl_algobase.h:71:0,
                 from /usr/include/c++/5/algorithm:61,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:28:
/usr/include/c++/5/bits/predefined_ops.h:194:17: note:   mismatched types ‘const std::complex<_Tp>’ and ‘const int’
  { return *__it == _M_value; }
                 ^
In file included from /usr/include/boost/config/no_tr1/complex.hpp:21:0,
                 from /usr/include/boost/math/policies/error_handling.hpp:15,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/console.h:39,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/interface_manager.h:38,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/include/combined_robot_hw/combined_robot_hw.h:35,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:29:
/usr/include/c++/5/complex:463:5: note: candidate: template<class _Tp> bool std::operator==(const std::complex<_Tp>&, const _Tp&)
     operator==(const complex<_Tp>& __x, const _Tp& __y)
     ^
/usr/include/c++/5/complex:463:5: note:   template argument deduction/substitution failed:
In file included from /usr/include/c++/5/bits/stl_algobase.h:71:0,
                 from /usr/include/c++/5/algorithm:61,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:28:
/usr/include/c++/5/bits/predefined_ops.h:194:17: note:   ‘std::__cxx11::basic_string<char>’ is not derived from ‘const std::complex<_Tp>’
  { return *__it == _M_value; }
                 ^
In file included from /usr/include/boost/config/no_tr1/complex.hpp:21:0,
                 from /usr/include/boost/math/policies/error_handling.hpp:15,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/console.h:39,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/interface_manager.h:38,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/include/combined_robot_hw/combined_robot_hw.h:35,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:29:
/usr/include/c++/5/complex:458:5: note: candidate: template<class _Tp> bool std::operator==(const std::complex<_Tp>&, const std::complex<_Tp>&)
     operator==(const complex<_Tp>& __x, const complex<_Tp>& __y)
     ^
/usr/include/c++/5/complex:458:5: note:   template argument deduction/substitution failed:
In file included from /usr/include/c++/5/bits/stl_algobase.h:71:0,
                 from /usr/include/c++/5/algorithm:61,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:28:
/usr/include/c++/5/bits/predefined_ops.h:194:17: note:   ‘std::__cxx11::basic_string<char>’ is not derived from ‘const std::complex<_Tp>’
  { return *__it == _M_value; }
                 ^
In file included from /usr/include/c++/5/bits/locale_facets.h:48:0,
                 from /usr/include/c++/5/bits/basic_ios.h:37,
                 from /usr/include/c++/5/ios:44,
                 from /usr/include/c++/5/istream:38,
                 from /usr/include/c++/5/sstream:38,
                 from /opt/ros/kinetic/include/ros/console.h:38,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/interface_manager.h:38,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/include/combined_robot_hw/combined_robot_hw.h:35,
                 from /home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/combined_robot_hw/src/combined_robot_hw.cpp:29:
/usr/include/c++/5/bits/streambuf_iterator.h:204:5: note: candidate: template<class _CharT, class _Traits> bool std::operator==(const std::istreambuf_iterator<_CharT, _Traits>&, const std::istreambuf_iterator<_CharT, _Traits>&)
     operator==(const istreambuf_iterator<_CharT, _Traits>& __a,
     ^
/usr/include/c++/5/bits/streambuf_iterator.h:204:5: note:   template argument deduction/substitution failed:
[... 494 lines cut ...]
make[2]: *** [realtime_tools/CMakeFiles/realtime_tools.dir/src/realtime_clock.cpp.o] Error 1
CMakeFiles/Makefile2:6179: recipe for target 'realtime_tools/CMakeFiles/realtime_tools.dir/all' failed
make[1]: *** [realtime_tools/CMakeFiles/realtime_tools.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 21%] Built target turtlebot3_msgs_generate_messages_eus
[ 21%] Built target turtlebot3_msgs_generate_messages_lisp
/home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/transmission_interface/src/transmission_parser.cpp: In static member function ‘static bool transmission_interface::TransmissionParser::parse(const string&, std::vector<transmission_interface::TransmissionInfo>&)’:
/home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/transmission_interface/src/transmission_parser.cpp:56:28: error: ‘nullptr’ was not declared in this scope
   TiXmlElement *trans_it = nullptr;
                            ^
/home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/transmission_interface/src/transmission_parser.cpp: In static member function ‘static bool transmission_interface::TransmissionParser::parseJoints(TiXmlElement*, std::vector<transmission_interface::JointInfo>&)’:
/home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/transmission_interface/src/transmission_parser.cpp:126:28: error: ‘nullptr’ was not declared in this scope
   TiXmlElement *joint_it = nullptr;
                            ^
/home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/transmission_interface/src/transmission_parser.cpp: In static member function ‘static bool transmission_interface::TransmissionParser::parseActuators(TiXmlElement*, std::vector<transmission_interface::ActuatorInfo>&)’:
/home/ishan/AuE893Spring19_SharmaIshan/catkin_ws/src/ros_control/transmission_interface/src/transmission_parser.cpp:198:31: error: ‘nullptr’ was not declared in this scope
   TiXmlElement *actuator_it = nullptr;
                               ^
ros_control/transmission_interface/CMakeFiles/transmission_interface_parser.dir/build.make:62: recipe for target 'ros_control/transmission_interface/CMakeFiles/transmission_interface_parser.dir/src/transmission_parser.cpp.o' failed
make[2]: *** [ros_control/transmission_interface/CMakeFiles/transmission_interface_parser.dir/src/transmission_parser.cpp.o] Error 1
CMakeFiles/Makefile2:13242: recipe for target 'ros_control/transmission_interface/CMakeFiles/transmission_interface_parser.dir/all' failed
make[1]: *** [ros_control/transmission_interface/CMakeFiles/transmission_interface_parser.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

ishaniis, 2020-02-27:
I don't even want C++ capability for my ROS, I would need help to chalk this out.

gavanderhoorn, 2020-02-27:
I'm slightly confused: this is a repository in the ROS-Industrial program. This is not a general ROS support channel.

I don't see any output which mentions packages hosted in this repository, nor any other packages which I recognise as being part of ROS-Industrial.

Did you post this to the wrong issue tracker?

gavanderhoorn, 2020-02-27:
I'm going to assume this was posted to the wrong tracker and I'm going to close this.

Feel free to keep commenting and if it makes sense, we can re-open.

closed 2020-02-27 by gavanderhoorn