ros-industrial/industrial_core #253: Support for ROS2

gtan039, 2020-03-20:
Hi, May I know if there will porting of the  industrial_robot_simulator and industrial_robot_client  to ROS2 in the near future?

gavanderhoorn, 2020-04-06:
There are currently no plans to do a direct port of the packages you mention to ROS 2.

`industrial_robot_client` et al. have a number of design and implementation issues which would be good to tackle instead of doing a straight port.

And with the availability of more and more publically available external motion interfaces in industrial robotics these days, the value of something like `simple_message` and `industrial_robot_client` would also be something to discuss.

xzkeee, 2022-03-31:
So how do I control an industrial robot with ros2? Do I have to write a driver without industrial_robot_client, directly connect to ros2_control ? What about those existing drivers, such as abb, fanuc drivers, how do they control by ros2? Or simply they can't, I have to stick to ros1? Thanks!

BrettHemes, 2022-08-09:
I am coming back to ROS after some hiatus and have the same question(2).  @gavanderhoorn do you have a summary of where industrial stands wrt ROS2 support (and/or perhaps have a list of things that need attention)?

ppbrown, 2023-01-18:
For anyone who wants to put in some coding effort, I started up my porting fork.

https://github.com/ppbrown/ros2_industrial_core

ppbrown, 2023-09-13:
simplest thing seems to be to stick to ros1 for now.
The good news is, you can at least indefinitely use the ros1 infrastructure by leveraging the ros1 docker container.
ie:
ros:noetic-ros-base

robinsonmm, 2023-09-19:
Thanks to all those that have posted here on this topic. We are jointly going to more explicitly line out our strategy relative to drivers and comms with manipulators and in parallel update Industrial Robot Driver Specification. As Gijs above illuminated there are now more OEMs offering external motion interfaces, and now more extensible and versionable ways to support ROS 2 interoperability with those interfaces, be it via microROS (Yaskawa) or ros2_control (UR). We will try to provide the resources to enable OEMs to develop, deploy and maintain ROS 2 support, in a way that benefits both developers and OEMs and sustains over changes to ROS 2 and to OEM controllers.

Thanks for your continued interest and engagement on open source for industry.

ppbrown, 2023-09-19:
Thanks for the interesting update @robinsonmm 
One additional question for you though: You mention microROS and Yaskawa in the same context. but going to the microROS website, and the github at https://github.com/micro-ROS, I dont see Yaskawa mentioned on either of them.
Can you shed more light on that relationship?

robinsonmm, 2023-09-20:
micro-ROS is simply an extreme resource constrained DDS implementation for devices with constraints, such as a robot controller. In the case of Yaskawa they chose micro-ROS as their pathway to implement a ROS 2 node on their controller, a different way to provide ROS interoperability for their manipulators. Their repo is here: https://github.com/Yaskawa-Global/motoros2 

Ted Miller of Yaskawa recently presented the work at the ROS-I Americas Annual Meeting, which a recording is available (Ted's talk: https://youtu.be/N8b775nXRTA) on the ROS-I YouTube playlist from the event here: https://www.youtube.com/playlist?list=PLXUpEXjGC63xwoXyAVtBsspOgS-7sAdx1