ros-industrial/industrial_core #256: Noetic release

TheDelus, 2020-06-02:
Hi, is a release for ros-noetic planned for this package?

gavanderhoorn, 2020-06-02:
See my response here: https://github.com/gavanderhoorn/industrial_robot_status_controller/issues/5#issuecomment-637607611.

EvilPictureBook, 2020-06-26:
Would this https://github.com/orocos/orocos_kinematics_dynamics/issues/218 mean that kdl is no longer blocking?

If that's the case, I'd be happy to make the changes and submit a PR following http://wiki.ros.org/noetic/Migration#Use_Orocos_KDL_and_BFL_rosdep_keys

That way it would be ready to go once MoveIt is released for Noetic (would it need a different branch since it would break the current one?).

simonschmeisser, 2020-08-17:
@EvilPictureBook yes, please open a PR!

I also noticed that `industrial_trajectory_filters` does not compile due to a API change in MoveIt. We need to add this virtual function:
```
virtual void planning_request_adapter::PlanningRequestAdapter::initialize(const ros::NodeHandle&)
```

seanyen, 2020-08-24:
Just wondering, is there any plans to have a dedicate branch for `melodic` or `noetic`? Would it be welcome to consider Windows [port fixes](https://github.com/ros-industrial/industrial_core/pull/251) as it moves to newer releases?

tylerjw, 2020-09-30:
MoveIt is now released for Noetic, so this can go forward if that was what was blocking it.

fmessmer, 2021-01-14:
there seems to be a `noetic` compatibility pr even: https://github.com/ros-industrial/industrial_core/pull/258
what's the status of the noetic release?
any chance `simple_message` gets release as a partial release?

simonschmeisser, 2021-01-15:
I haven't received any feedback to my PR so far unfortunately ... maybe you could get it a review @fmessmer and then @gavanderhoorn would merge it?

fmessmer, 2021-01-15:
> review

I'm mainly interested in seeing `simple_message` released....I have no insight into these `industrial-core` packages as I'm only trying to satisfy our noetic dependencies...thus, unfortunately, I don't see myself qualified to review...but `simple_message` doesn't seem to be blocked as it already seems `noetic`-ready...thus, I asked for a partial release...sorry

fmessmer, 2021-01-19:
I can confirm that (at least) `simple_message` compiles on noetic from `ros-industrial/industrial_core#kinetic-devel`
:point_right: so a partial release should be possible for `simple_message`

I can also confirm that with #258 **all** packages of `industrial_core` compile on noetic
:point_right: so a full release should be possible after merging #258

gavanderhoorn, 2021-06-28:
Done: https://github.com/ros/rosdistro/pull/30044.

Even though it hasn't been merged yet, I'm going to close this issue, as it's basically just 'formalities' at this point.

gavanderhoorn, 2021-06-28:
Note: I opted to release without #271 merged. It'll already unblock releases of other packages and as that PR potentially changes behaviour, I thought it'd be good to first get a release out.

closed 2021-06-28 by gavanderhoorn