ros-industrial/industrial_core #257: Enable debug messages of industrial_robot_client

jediofgever, 2020-08-11:
Hi, 
I would like to see all debug messages pushed to /rosout from nodes that are in `industrial_robot_client`
I tried following but I still cannot see any debug messages from nodes of `industrial_robot_client`.

I installed `ros-melodic-industrial-utils-dbgsym` and built the project with debug messages enabled;
`catkin_make -DCMAKE_BUILD_TYPE=Debug install ` 

finally I add  `output="screen" ` to launch file where we make node call. 

despite these I am unable to visualize debug messages. I would be glad if you  can let me know what is required in order to enable debug messages.

Thank you for your time.

gavanderhoorn, 2020-08-11:
> I would like to see all debug messages pushed to /rosout from nodes that are in `industrial_robot_client`

For that you'd have to change the log level of the nodes, while they are running. You could use [rqt_logger_level](http://wiki.ros.org/rqt_logger_level).

> I installed `ros-melodic-industrial-utils-dbgsym` and built the project with debug messages enabled;
> `catkin_make -DCMAKE_BUILD_TYPE=Debug install `

This will not change the logger levels. It will only build the binaries with debug symbols embedded, which is what you'd do if you'd like to use GDB (or other debuggers).

> finally I add `output="screen" ` to launch file where we make node call.

That would be necessary if you'd like to see log messages in the terminal. If you don't have this configured even `rqt_logger_level` won't help.

gavanderhoorn, 2020-08-11:
As this is not an actual issue with the packages in this repository, I'm going to close it.

Feel free to keep commenting on the issue of course.

closed 2020-08-11 by gavanderhoorn

jediofgever, 2020-08-11:
Thanks @gavanderhoorn for your help. Yes not really an issue with the industrial_core itself but I thought I will get a response faster which turns out to be true 🙏

gavanderhoorn, 2020-08-11:
Please post these kind of questions on ROS Answers in the future.

I haven't checked, but I'm pretty sure there are already N duplicates of your question (ie: *how to change log level of a node*).

gavanderhoorn, 2020-08-11:
Not sure what happened, but your last comment got posted 6 times. I've deleted the duplicates.

jediofgever, 2020-08-11:
Yes my bad , I am using GitHub mobile app, must be some issue my connection I guess. 
You are right about the availability of answer on changing log level, it should be available in ros answers. Will keep your suggestion in mind for next time.