ros-industrial/industrial_core #274: Build issue breaking motoman/fanuc/internal drivers

akashjinandra, 2021-07-14:
Hello, I'm running Ubuntu 18.04 with ROS Melodic, I just starting seeing an issue where the simple message/industrial core is causing issues in all of my motoman, fanuc and internal arm(ros industrial driver based) drivers. 

Here is my output: In file included from /opt/ros/melodic/include/simple_message/socket/tcp_socket.h:36:0,
                 from /opt/ros/melodic/include/simple_message/socket/tcp_client.h:36,
                 from /opt/ros/melodic/include/industrial_robot_client/joint_trajectory_interface.h:44,
                 from /opt/ros/melodic/include/industrial_robot_client/joint_trajectory_streamer.h:36,
                 from /home/akash/test_ws/src/fanuc/fanuc_driver/src/fanuc_joint_streamer_node.cpp:41:
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:196:3: error: ‘sockaddr_in’ does not name a type; did you mean ‘caddr_t’?
   sockaddr_in sockaddr_;
   ^~~~~~~~~~~
   caddr_t
/opt/ros/melodic/include/simple_message/socket/simple_socket.h: In constructor ‘industrial::simple_socket::SimpleSocket::SimpleSocket()’:
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:150:19: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’; did you mean ‘sock_handle_’?
     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
                   ^~~~~~~~~
                   sock_handle_
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:150:46: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’; did you mean ‘sock_handle_’?
     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
                                              ^~~~~~~~~
                                              sock_handle_
make[2]: *** [CMakeFiles/fanuc_driver_motion_streaming_interface_bswap.dir/src/fanuc_joint_streamer_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/fanuc_driver_motion_streaming_interface_bswap.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /opt/ros/melodic/include/simple_message/socket/tcp_socket.h:36:0,
                 from /opt/ros/melodic/include/simple_message/socket/tcp_client.h:36,
                 from /opt/ros/melodic/include/industrial_robot_client/robot_state_interface.h:41,
                 from /home/akash/test_ws/src/fanuc/fanuc_driver/src/fanuc_robot_state_node.cpp:41:
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:196:3: error: ‘sockaddr_in’ does not name a type; did you mean ‘caddr_t’?
   sockaddr_in sockaddr_;
   ^~~~~~~~~~~
   caddr_t
/opt/ros/melodic/include/simple_message/socket/simple_socket.h: In constructor ‘industrial::simple_socket::SimpleSocket::SimpleSocket()’:
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:150:19: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’; did you mean ‘sock_handle_’?
     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
                   ^~~~~~~~~
                   sock_handle_
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:150:46: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’; did you mean ‘sock_handle_’?
     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
                                              ^~~~~~~~~
                                              sock_handle_
make[2]: *** [CMakeFiles/fanuc_driver_robot_state.dir/src/fanuc_robot_state_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/fanuc_driver_robot_state.dir/all] Error 2
In file included from /opt/ros/melodic/include/simple_message/socket/tcp_socket.h:36:0,
                 from /opt/ros/melodic/include/simple_message/socket/tcp_client.h:36,
                 from /opt/ros/melodic/include/industrial_robot_client/robot_state_interface.h:41,
                 from /home/akash/test_ws/src/fanuc/fanuc_driver/src/fanuc_robot_state_node.cpp:41:
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:196:3: error: ‘sockaddr_in’ does not name a type; did you mean ‘caddr_t’?
   sockaddr_in sockaddr_;
   ^~~~~~~~~~~
   caddr_t
/opt/ros/melodic/include/simple_message/socket/simple_socket.h: In constructor ‘industrial::simple_socket::SimpleSocket::SimpleSocket()’:
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:150:19: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’; did you mean ‘sock_handle_’?
     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
                   ^~~~~~~~~
                   sock_handle_
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:150:46: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’; did you mean ‘sock_handle_’?
     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
                                              ^~~~~~~~~
                                              sock_handle_
make[2]: *** [CMakeFiles/fanuc_driver_robot_state_bswap.dir/src/fanuc_robot_state_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/fanuc_driver_robot_state_bswap.dir/all] Error 2
In file included from /opt/ros/melodic/include/simple_message/socket/tcp_socket.h:36:0,
                 from /opt/ros/melodic/include/simple_message/socket/tcp_client.h:36,
                 from /opt/ros/melodic/include/industrial_robot_client/joint_trajectory_interface.h:44,
                 from /opt/ros/melodic/include/industrial_robot_client/joint_trajectory_streamer.h:36,
                 from /home/akash/test_ws/src/fanuc/fanuc_driver/src/fanuc_joint_streamer_node.cpp:41:
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:196:3: error: ‘sockaddr_in’ does not name a type; did you mean ‘caddr_t’?
   sockaddr_in sockaddr_;
   ^~~~~~~~~~~
   caddr_t
/opt/ros/melodic/include/simple_message/socket/simple_socket.h: In constructor ‘industrial::simple_socket::SimpleSocket::SimpleSocket()’:
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:150:19: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’; did you mean ‘sock_handle_’?
     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
                   ^~~~~~~~~
                   sock_handle_
/opt/ros/melodic/include/simple_message/socket/simple_socket.h:150:46: error: ‘class industrial::simple_socket::SimpleSocket’ has no member named ‘sockaddr_’; did you mean ‘sock_handle_’?
     memset(&this->sockaddr_, 0, sizeof(this->sockaddr_));
                                              ^~~~~~~~~
                                              sock_handle_
make[2]: *** [CMakeFiles/fanuc_driver_motion_streaming_interface.dir/src/fanuc_joint_streamer_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/fanuc_driver_motion_streaming_interface.dir/all] Error 2

Sorry for the big file dump, I just am not sure what I could do to fix this. I tried getting ros industrial from source on melodic devel and it still fails.

gavanderhoorn, 2021-07-15:
This is caused by #262.

This requires some updates in dependents. See https://github.com/ros-industrial/fanuc/pull/324 for an example PR. Specifically https://github.com/ros-industrial/fanuc/pull/324/commits/37a13af73cfece4c52a26e1b8ff6a7b7bb68805e.

We might actually revert that PR and reintroduce it in a `noetic-devel` branch.

gavanderhoorn, 2021-07-15:
The reason we can't just do something like `add_definitions(..)` in an extras file is because `simple_message` and `industrial_core` basically export three different libraries, with two variants each (swap, non-swap).

So dependents must use the correct set of defines, otherwise they would be out-of-sync with whatever version of `simple_message` and `industrial_robot_client` they are linking against.

gavanderhoorn, 2021-07-15:
Note: I've submitted https://github.com/ros/rosdistro/pull/30228, which reverts the last release on Melodic.

The buildfarm will build the old version of the packages and everyone depending on the binaries will be unbroken.

Then we'll revert #262 in `melodic-devel`, re-release `melodic` without #262.

I hate branching for different ROS versions, but in this case it doesn't seem like we have a choice.

gavanderhoorn, 2021-07-15:
Ok, instead of reverting everything, I've added some bw-compatibility to the changes that were introduced in #262, which should make things build again on systems without something like https://github.com/ros-industrial/fanuc/pull/324.

@akashjinandra: could you check whether https://github.com/ros-industrial/industrial_core/compare/melodic-devel...gavanderhoorn:add_bw_compat_262 fixes your builds?

It would still be better to migrate to use `simple_message_DEFINITIONS` et al., but we shouldn't break packages not under our control any more like this in `melodic`.

PR #275 (2021-07-15): Add bw-compatibility to simple_message ifdefs
For now (ie: in `melodic` and `melodic-devel`), accept the old `#define`s `ROS` and `LINUXSOCKETS` in addition to the newer ones (ie: the namespaced ones).

Should fix #274 and fix builds which haven't been updated to use the new `simple_message_DEFINITIONS`.

This was already suggested actually by @jdlangs in #190, but I removed it, hoping to get away with it in Melodic.

Unfortunately that didn't work, so here we are.
closed 2021-07-15 by gavanderhoorn