ros-industrial/industrial_core #278: - Portability on ROS1 Noetic.

giordy-sudo, 2021-12-16:
Hello, I have some issues when I compile (catkin build) these packages : 
- **industrial_robot_client** ;
- **industrial_utils.**

In particular errors are : 
- industrial_robot_client : 

[image]
[image]

- industrial_utils : 

[image]

Thanks a lot.

gavanderhoorn, 2021-12-16:
You appear to use a fork of `industrial_core`.

All `industrial_core` packages have been released for ROS Noetic, and CI runs regularly (see [here](https://github.com/ros-industrial/industrial_core/actions/runs/1586875438) for the latest run on Ubuntu Focal).

It's likely the fork is outdated.

gavanderhoorn, 2021-12-16:
Closing as this is not an issue with the packages in this repository.

gavanderhoorn, 2021-12-16:
PS: and please, do not post screenshots of terminals.

It's all text, so copy-paste the errors as text into your github issue.

Do make sure to use [fenced codeblocks](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks) though.

closed 2021-12-16 by gavanderhoorn

giordy-sudo, 2021-12-16:
Hello, thanks. Also withe the modifications that you sent me I still have the same errors. 
I apply these changes : https://github.com/ros-industrial/industrial_core/commit/ea3d805e1271550be312409d6474177e67796e15.

gavanderhoorn, 2021-12-16:
I'm not sure I understand.

I have not sent you any modifications.

> I apply these changes : [ea3d805](https://github.com/ros-industrial/industrial_core/commit/ea3d805e1271550be312409d6474177e67796e15).

`cherry-pick`ing a single commit will not solve the problem.

giordy-sudo, 2021-12-16:
Ok sorry can you explain in a simple way how can I solve my problem ?
Thanks.

gavanderhoorn, 2021-12-16:
I would suggest to ask the maintainers of the fork you are trying to use.

I don't know what their motivation for forking was, and I don't know what changes they've made to the code.

At a high level, you'd have to merge the differences between their fork and upstream (ie: this repository).

giordy-sudo, 2021-12-16:
So who are the maintainers of the fork I'm trying to use ?

gavanderhoorn, 2021-12-16:
I have no idea.

Wouldn't you be the one who knows what he's `git clone`-ing?